MichaelZaidman/hid-ft260 #1: Doesn't automatically load 'i2c-dev'

mungewell, 2021-03-14:
Hi,
Was able to build the kernel module, but it looks like it's missing a dependency as it doesn't automatically load 'i2c-dev.ko'. Maybe this is by design, but it tripped me up for a bit.

Without this I was not able to see any I2C interfaces listed under '/dev/', but they did exist under '/sys/bus/i2c'.

Once loaded I DID see the interface and was able to access chip via Python. Thanks,
Simon.

```
root@thevoid:/home/simon# cat i2c_test.py
import smbus

bus = smbus.SMBus(20)

DEVICE_ADD = 0x55
DEVICE_REG = 0xE0

version = bus.read_i2c_block_data(DEVICE_ADD, DEVICE_REG, 4)
print(version)

root@thevoid:/home/simon# python3 i2c_test.py 
[0, 0, 0, 1]
```

MichaelZaidman, 2021-03-15:
Hi Simon,

Thanks for testing!

The driver works on Ubuntu 16.04 and 20.04 with Ubuntu stock kernel and the 5.11 kernel in Virtual Box VM without explicit dependencies. No need to load i2c-dev.ko since it’s compiled into the kernel.

What Linux distribution do you use?

Michael

mungewell, 2021-03-16:
I'm using Xubuntu 18.04 which should be same kernel as Ubuntu, however as stated I have rebuilt kernel from scratch.
```
$ uname -a
Linux thevoid 5.12.0-rc2+ #8 SMP Sat Mar 13 20:02:06 MST 2021 x86_64 x86_64 x86_64 GNU/Linux
```

For my latest kernels this is a 'm', but one of them has it as 'y'
```
$ grep CONFIG_I2C_CHARDEV /boot/config-*           
/boot/config-4.15.0-136-generic:CONFIG_I2C_CHARDEV=y
/boot/config-5.10.0-051000rc4-generic:CONFIG_I2C_CHARDEV=m
/boot/config-5.12.0-rc2+:CONFIG_I2C_CHARDEV=m
```

Oddly it looks like I2C_CHARDEV doesn't have a default setting...
https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/i2c/Kconfig?h=v5.12-rc3#n51

MichaelZaidman, 2021-03-16:
I am not using Xubuntu, but it sounds like they configure their kernels differently.

As mentioned earlier, all Ubuntu stock kernels I tested have the i2c-dev built-in. From my Ubuntu VMs:
```
michael@michael-VirtualBox:~$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 16.04.7 LTS
Release:	16.04
Codename:	xenial
michael@michael-VirtualBox:~$ grep CONFIG_I2C_CHARDEV /boot/config-*
/boot/config-4.15.0-132-generic:CONFIG_I2C_CHARDEV=y
/boot/config-4.15.0-133-generic:CONFIG_I2C_CHARDEV=y
/boot/config-5.11.0-051100rc7-generic:CONFIG_I2C_CHARDEV=y

michael@michael-VirtualBox:/$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.2 LTS
Release:	20.04
Codename:	focal
michael@michael-VirtualBox:/$ grep CONFIG_I2C_CHARDEV /boot/config-*
/boot/config-5.10.15-051015-generic:CONFIG_I2C_CHARDEV=y
/boot/config-5.11.0-051100-generic:CONFIG_I2C_CHARDEV=y
/boot/config-5.8.0-43-generic:CONFIG_I2C_CHARDEV=y
/boot/config-5.8.0-44-generic:CONFIG_I2C_CHARDEV=y
```
Nevertheless, you raise a good point and, I am going to add an explicit dependency on the i2c-dev.ko later.

Do you mind replying with “tested-by” to the patch in the kernel mailing lists? Or, simply send me your e-mail, and I will add your “tested-by” to the next version of the patch.

MichaelZaidman, 2022-04-07:
HOWTO define a dependency on the i2c-dev [https://www.ddcutil.com/kernel_module/](https://www.ddcutil.com/kernel_module/)

closed 2022-04-07 by MichaelZaidman