MichaelZaidman/hid-ft260 #9: Setting gpio2_fun

hmohajeri, 2023-04-16:
Hi, is there a way to set gpio2_func without "sudo" ? I mean could the file come up as +666 from the start.
sudo bash -c "echo 0 > $SYSFS_FT260/gpio2_func"

Regards,

MichaelZaidman, 2023-04-22:
Hi @hmohajeri,
```
include/linux/device.h
    #define DEVICE_ATTR_RW(_name) \
	struct device_attribute dev_attr_##_name = __ATTR_RW(_name)

include/linux/sysfs.h
    #define __ATTR_RW(_name) __ATTR(_name, 0644, _name##_show, _name##_store)  
```
You might have already seen a similar question - [https://stackoverflow.com/questions/40776170/cannot-set-attribute-permissions-to-0666-in-sysfs](https://stackoverflow.com/questions/40776170/cannot-set-attribute-permissions-to-0666-in-sysfs). It is referring to an outdated kernel version, but the explanation is correct - only the root should write to /sys/kernel files.

-Michael

hmohajeri, 2023-04-25:
Many thanks and appreciate the feedback.

regards,

closed 2023-04-25 by hmohajeri