MichaelZaidman/hid-ft260 #10: irq DIO8

sora1874, 2023-05-09:
if only use i2c interface, the irq dio8 can get report？

MichaelZaidman, 2023-05-09:
The IRQ line via GPIO8 will not generate HID report. It notifies the FT260 to query the data from the device to be sent to the host via USB when the FT260 is configured to work in the HID-over-I2C mode.

sora1874, 2023-05-11:
[image]
When setting up the uart interface , can receive 0xb1 report message.
Add code to open interrupt, like this: 
		/* init gpio irq */
		dev->irqchip.name = "ft260_irq";
		dev->irqchip.irq_startup = ft260_gpio_irq_startup;
		dev->irqchip.irq_shutdown = ft260_gpio_irq_shutdown;
		dev->irqchip.irq_mask = ft260_gpio_irq_mask;
		dev->irqchip.irq_unmask = ft260_gpio_irq_unmask;
		dev->irqchip.irq_set_type = ft260_gpio_irq_type;
		dev->irqchip.flags = IRQCHIP_MASK_ON_SUSPEND;
		girq = &dev->gc->irq;
		girq->chip = &dev->irqchip;
		girq->parent_handler = NULL;
		girq->num_parents = 0;
		girq->parents = NULL;
		girq->default_type = IRQ_TYPE_NONE;
		girq->handler = handle_simple_irq;
when get 0xb1 message, add code in ft260_raw_event:
              else if (xfer->report == 0xb1) {
                     irq = dev->gc->to_irq(dev->gc,  3);
                    handle_nested_irq(irq);
             }
These modifications may be problematic. Do you have a demo for the GPIO IRQ section

closed 2023-05-11 by sora1874