MichaelZaidman/hid-ft260 #16: GPIO doesn't work under RaspberryPi-OS Bookworm,

jk-advedes, 2024-02-20:
Dear Michael,
while HID to i2c still works fine, GPIO stopped to work. Outcome of "gpioinfo":
gpiochip2 - 14 lines:
	line   0:      unnamed       unused   input  active-high 
	line   1:      unnamed       unused   input  active-high 
	line   2:      unnamed       unused   input  active-high 
	line   3:      unnamed       unused   input  active-high 
	line   4:      unnamed       unused   input  active-high 
	line   5:      unnamed       unused   input  active-high 
	line   6:      unnamed       unused   input  active-high 
	line   7:      unnamed       unused   input  active-high 
	line   8:      unnamed       unused   input  active-high 
	line   9:      unnamed       unused   input  active-high 
	line  10:      unnamed       unused   input  active-high 
	line  11:      unnamed       unused   input  active-high 
	line  12:      unnamed       unused   input  active-high 
	line  13:      unnamed       unused   input  active-high 

As well, shell commands from example on github page stopped to work:
$ ~/hid-ft260 $ cat $SYSFS_FT260/gpio2_func
cat: /gpio2_func: No such file or directory

Confirmation that kernel module is correctly loaded:
$ i2cdetect -l
i2c-20	i2c       	fef04500.i2c                    	I2C adapter
i2c-21	i2c       	fef09500.i2c                    	I2C adapter
i2c-22	i2c       	FT260 usb-i2c bridge            	I2C adapter

Would you be so kind and test yourself to confirm please? Thanks in advance!

jk-advedes, 2024-02-28:
Hello Michael, many thanks for your interest in the issue! Answers per each question:
1)
tested on: RPI4
$ uname -a
Linux rpi4 6.1.0-rpi8-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.73-1+rpt1 (2024-01-25) aarch64 GNU/Linux
$ git show
commit 4bbf04234974f98605d8d6ab07c94d6b8a18e45d
$ make V=1
make -C /lib/modules/6.1.0-rpi8-rpi-v8/build M=/home/jura/hid-ft260 modules
make[1]: Entering directory '/usr/src/linux-headers-6.1.0-rpi8-rpi-v8'
make -f /usr/src/linux-headers-6.1.0-rpi8-common-rpi/scripts/Makefile.build obj=/home/jura/hid-ft260 need-builtin=1 need-modorder=1 
   gcc-12 -Wp,-MMD

2)
$ SYSFS_FT260=$(d=$(ls /sys/bus/hid/drivers/ft260 | grep '0403:6030'); echo /sys/bus/hid/drivers/ft260/$d)
$ echo $SYSFS_FT260
/sys/bus/hid/drivers/ft260/0003:0403:6030.0001 0003:0403:6030.0002
$ ls $SYSFS_FT260
ls: cannot access '0003:0403:6030.0002': No such file or directory
'/sys/bus/hid/drivers/ft260/0003:0403:6030.0001':
chip_mode  gpio        gpiog_func       i2c_reset        pwren_status       uart_mode
clock      gpio2_func  hid_over_i2c_en  modalias         report_descriptor  uevent
clock_ctl  gpioa_func  i2c-22           power            subsystem
driver     gpiochip2   i2c_enable       power_saving_en  suspend_status
$ ls /sys/bus/hid/drivers/ft260/0003:0403:6030.0002
driver  modalias  power  report_descriptor  subsystem  uevent

3)
$ i2cdetect -l
i2c-20	i2c       	fef04500.i2c                    	I2C adapter
i2c-21	i2c       	fef09500.i2c                    	I2C adapter
i2c-22	i2c       	FT260 usb-i2c bridge            	I2C adapter

Note: FT260 is pin-strapped DCNP0 = high, DCNF1 = low (see datasheet section 5.1 Device Interface Configuration).

MichaelZaidman, 2024-02-28:
Hi @jk-advedes,

I see, the SYSFS_FT260 variable is broken. Please try the `SYSFS_FT260=/sys/bus/hid/drivers/ft260/0003:0403:6030.0001` command, and then, you should be able to access the ft260 sysfs attributes like `cat $SYSFS_FT260/gpio2_func`

Please let me know if it works for you.

MichaelZaidman, 2024-03-01:
> While HID to i2c still works fine, GPIO stopped working.

The gpio works fine. The issue was with the sysfs pass discovered by the helper script from README. It sets the environment variable correctly with a single interface but not when both are enabled. Since FT260 is a USB device, it appears under different sysfs paths on other systems. So, the user is expected to figure out the correct path by himself. To simplify this task, I provided the oneliner bash script, assuming only a single i2c interface is enabled since the driver in the main branch does not support the second interface yet. Nevertheless, I added the setenv.sh script supporting multiple interfaces and FT260 devices, and updated the README with the example. Please try. I would appreciate your feedback.

closed 2024-06-08 by MichaelZaidman