MichaelZaidman/hid-ft260 #18: ttyFTx not  showing up

DevDebugger, 2024-04-17:
I installed the driver, insmod the .ko file , ran the setenv script, however i am not able to see any ttyFTx device in /dev/. Am i missing a step?

MichaelZaidman, 2024-04-17:
The UART support is available only in the separate uart branch and has not yet been merged into the main branch. However, please note that the uart branch does not include the GPIO support found in the main branch. If you don't need the GPIO support, you can use the uart branch. On the other hand, if you require both UART and GPIO support, I have created a local branch that includes UART support on top of GPIO support. I will upload this branch in the next few days.

DevDebugger, 2024-04-23:
thank you! was this branch every uploaded?

MichaelZaidman, 2024-04-24:
Yes, I merged it into the main branch right now.

MichaelZaidman, 2024-04-30:
Have you had a chance to try it? Can we close the ticket?

closed 2024-06-08 by MichaelZaidman