MichaelZaidman/hid-ft260 #20: I2C_SMBUS_QUICK support

wayne-tung, 2024-04-30:
Hi, all

This ft260 driver is very helpful for me.

I have one question here,

I found the commit that you removed the **I2C_SMBUS_QUICK** commands.

Can we implement **SMBUS quick write** command by **ft260_i2c_write**?
If we set **ft260_i2c_write** parameter len to 0, is that okay?

MichaelZaidman, 2024-04-30:
Hi @wayne-tung,

The reason for the I2C_SMBUS_QUICK removal is explained in the https://github.com/MichaelZaidman/hid-ft260/commit/e8d65cc8c6054dd2241382b5ffef8d07419266e7 commit message.

The I2C_SMBUS_QUICK sends a single bit to the device at the place of the Rd/Wr bit.
But the FT260 does not support 1-bit writes. The len=0 does not help. It still sends a command byte since it's the SMB protocol. The i2cdetect utility uses the I2C_SMBUS_QUICK to probe the devices on the bus by default. With len=0, it writes 0 into the first address of the devices on the bus, which can cause malfunctions. For instance, in the case of an I2C mux, it may cause it to forget its configuration.

wayne-tung, 2024-04-30:
Thanks for your reply.

Why ask for I2C_SMBUS_QUICK is that.
I found that when I'm trying to insert the ft260 driver, it will take about 40 seconds to complete.

After I study and trace the problem,
I found that some i2c drivers will register address_list for example: **lm75** below.

`/* Addresses scanned */
static const unsigned short normal_i2c[] = { 0x48, 0x49, 0x4a, 0x4b, 0x4c,
					0x4d, 0x4e, 0x4f, I2C_CLIENT_END };
static struct i2c_driver lm75_driver = {
	.class		= I2C_CLASS_HWMON,
	.driver = {
		.name	= "lm75",
		.of_match_table = of_match_ptr(lm75_of_match),
		.pm	= LM75_DEV_PM_OPS,
	},
	.probe		= lm75_probe,
	.id_table	= lm75_ids,
	.detect		= lm75_detect,
	.address_list	= normal_i2c,
};`

**i2c_add_adapter** function will scan all address list.
And because we don't have **smbus_quick** function, i2c-core will use **i2c read** function to scan the addresses.

In **ft260_i2c_read** function we wait until raw_event coming for 5 seconds.
`		timeout = msecs_to_jiffies(5000);
		if (!wait_for_completion_timeout(&dev->wait, timeout)) {
			ret = -ETIMEDOUT;
			ft260_i2c_reset(hdev);
			goto ft260_i2c_read_exit;
		}`

I guess because ft260 driver is still in probe function so the driver cannot receive hid raw_event yet.
I will cause for each address takes 5 seconds wait for timeout.

Do you have any idea about this issue?

MichaelZaidman, 2024-04-30:
I conducted a quick experiment using ft260_i2c_write to enable I2C_SMBUS_QUICK instead of ft260_smbus_write and passing len=0, but it did not work as expected. 

```diff --git a/hid-ft260.c b/hid-ft260.c
index b2c6a50..8361138 100644
--- a/hid-ft260.c
+++ b/hid-ft260.c
@@ -633,8 +633,8 @@ static int ft260_i2c_write(struct ft260_device *dev, u8 addr, u8 *data,
        struct ft260_i2c_write_request_report *rep =
                (struct ft260_i2c_write_request_report *)dev->i2c_wr_buf;
 
-       if (len < 1)
-               return -EINVAL;
+       //if (len < 1)
+       //      return -EINVAL;
 
        rep->flag = FT260_FLAG_START;
 
@@ -647,7 +647,7 @@ static int ft260_i2c_write(struct ft260_device *dev, u8 addr, u8 *data,
                        wr_len = FT260_WR_I2C_DATA_MAX;
                }
 
-               rep->report = FT260_I2C_DATA_REPORT_ID(wr_len);
+               rep->report = FT260_I2C_DATA_REPORT_ID(1);
                rep->address = addr;
                rep->length = wr_len;
 
@@ -868,6 +868,7 @@ static int ft260_smbus_xfer(struct i2c_adapter *adapter, u16 addr, u16 flags,
        int ret;
        struct ft260_device *dev = i2c_get_adapdata(adapter);
        struct hid_device *hdev = dev->hdev;
+       unsigned char byte = 1;
 
        ft260_dbg("smbus size %d\n", size);
 
@@ -881,6 +882,15 @@ static int ft260_smbus_xfer(struct i2c_adapter *adapter, u16 addr, u16 flags,
        }
 
        switch (size) {
+       case I2C_SMBUS_QUICK:
+               if (read_write == I2C_SMBUS_READ)
+                       ret = ft260_i2c_read(dev, addr, &data->byte, 0,
+                                            FT260_FLAG_START_STOP);
+               else
+                       ret = ft260_i2c_write(dev, addr, &byte, 0,
+                                             FT260_FLAG_START_STOP);
+               break;
+
        case I2C_SMBUS_BYTE:
                if (read_write == I2C_SMBUS_READ)
                        ret = ft260_i2c_read(dev, addr, &data->byte, 1,
@@ -963,7 +973,7 @@ smbus_exit:
 
 static u32 ft260_functionality(struct i2c_adapter *adap)
 {
-       return I2C_FUNC_I2C | I2C_FUNC_SMBUS_BYTE |
+       return I2C_FUNC_I2C | I2C_FUNC_SMBUS_BYTE | I2C_FUNC_SMBUS_QUICK |
               I2C_FUNC_SMBUS_BYTE_DATA | I2C_FUNC_SMBUS_WORD_DATA |
               I2C_FUNC_SMBUS_BLOCK_DATA | I2C_FUNC_SMBUS_I2C_BLOCK;
 }
```

I have no LM75 on the bus at the moment to try to reproduce the issue, but when I invoke the i2cdetect with the -r flag, it scans the bus with ft260_i2c_read which exits immediately with an i2c bus error 0x26 and does not wait for a timeout expiration. So, the i2cdetect scans the entire bus in a fraction of a second.

```
$ sudo i2cdetect -y -r 0
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: 50 51 -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- -- 
```
dmsg fragment around 50, 51 addresses
```
[558408.146571] ft260_smbus_xfer: smbus size 1
[558408.146573] ft260_i2c_read: rep 0xc2 addr 0x4e len 1 rlen 1 flag 0x6
[558408.148117] ft260_raw_event: i2c resp: rep 0xd0 len 1
[558408.148316] ft260_xfer_status: bus_status 0x26, clock 100
[558408.148318] ft260 0003:0403:6030.0010: i2c bus error: 0x26
[558408.148551] ft260_i2c_reset: done
[558408.148562] ft260_smbus_xfer: smbus size 1
[558408.148563] ft260_i2c_read: rep 0xc2 addr 0x4f len 1 rlen 1 flag 0x6
[558408.150106] ft260_raw_event: i2c resp: rep 0xd0 len 1
[558408.150286] ft260_xfer_status: bus_status 0x26, clock 100
[558408.150287] ft260 0003:0403:6030.0010: i2c bus error: 0x26
[558408.150475] ft260_i2c_reset: done
[558408.150487] ft260_smbus_xfer: smbus size 1
[558408.150489] ft260_i2c_read: rep 0xc2 addr 0x50 len 1 rlen 1 flag 0x6
[558408.152114] ft260_raw_event: i2c resp: rep 0xd0 len 1
[558408.152263] ft260_xfer_status: bus_status 0x20, clock 100
[558408.152269] ft260_smbus_xfer: smbus size 1
[558408.152270] ft260_i2c_read: rep 0xc2 addr 0x51 len 1 rlen 1 flag 0x6
[558408.154119] ft260_raw_event: i2c resp: rep 0xd0 len 1
[558408.154348] ft260_xfer_status: bus_status 0x20, clock 100
[558408.154362] ft260_smbus_xfer: smbus size 1
[558408.154374] ft260_i2c_read: rep 0xc2 addr 0x52 len 1 rlen 1 flag 0x6
[558408.156127] ft260_raw_event: i2c resp: rep 0xd0 len 1
[558408.156346] ft260_xfer_status: bus_status 0x26, clock 100
[558408.156349] ft260 0003:0403:6030.0010: i2c bus error: 0x26
[558408.156586] ft260_i2c_reset: done
[558408.156627] ft260_smbus_xfer: smbus size 1
[558408.156630] ft260_i2c_read: rep 0xc2 addr 0x53 len 1 rlen 1 flag 0x6
[558408.159111] ft260_raw_event: i2c resp: rep 0xd0 len 1
[558408.159351] ft260_xfer_status: bus_status 0x26, clock 100
[558408.159356] ft260 0003:0403:6030.0010: i2c bus error: 0x26
[558408.159599] ft260_i2c_reset: done
[558408.159648] ft260_smbus_xfer: smbus size 1
[558408.159651] ft260_i2c_read: rep 0xc2 addr 0x54 len 1 rlen 1 flag 0x6
[558408.162108] ft260_raw_event: i2c resp: rep 0xd0 len 1
[558408.162303] ft260_xfer_status: bus_status 0x26, clock 100
[558408.162309] ft260 0003:0403:6030.0010: i2c bus error: 0x26
[558408.162548] ft260_i2c_reset: done
[558408.162570] ft260_smbus_xfer: smbus size 1
[558408.162572] ft260_i2c_read: rep 0xc2 addr 0x55 len 1 rlen 1 flag 0x6
[558408.164105] ft260_raw_event: i2c resp: rep 0xd0 len 1
[558408.164318] ft260_xfer_status: bus_status 0x26, clock 100
[558408.164323] ft260 0003:0403:6030.0010: i2c bus error: 0x26
[558408.164558] ft260_i2c_reset: done
```

wayne-tung, 2024-05-01:
To reproduce the issue, you don't need to have real lm75 device.
Only need to insert lm75 driver.

modprobe lm75
modprobe hid-ft260

and then if you have ft260 on your board.
you will wait for a long time in command **modprobe hid-ft260**

MichaelZaidman, 2024-05-01:
Hi @wayne-tung, thanks for the clarification. I was able to reproduce the issue.

I fixed it in this commit https://github.com/MichaelZaidman/hid-ft260/commit/07893884aeeee36a8be43f5c30220ce9f78e4865, reducing the scan time to ~250ms.

```[611181.186203] hid-generic 0003:0403:6030.0010: hidraw2: USB HID v1.11 Device [FTDI FT260] on usb-0000:00:14.0-3.4/input0
[611181.186681] hid-generic 0003:0403:6030.0011: hidraw4: USB HID v1.11 Device [FTDI FT260] on usb-0000:00:14.0-3.4/input1
[611190.672365] ft260 0003:0403:6030.0010: chip code: 0260 0200
[611190.672661] ft260_get_interface_type: interface:  0x00
[611190.672665] ft260_get_interface_type: chip mode:  0x03
[611190.672666] ft260_get_interface_type: clock_ctl:  0x02
[611190.672667] ft260_get_interface_type: i2c_enable: 0x01
[611190.672668] ft260_get_interface_type: uart_mode:  0x04
[611190.672670] ft260_get_interface_type: gpio2_func: 0x01
[611190.672671] ft260_get_interface_type: gpioA_func: 0x03
[611190.672672] ft260_get_interface_type: gpioG_func: 0x06
[611190.672673] ft260_get_interface_type: wakeup_int: 0x01
[611190.672674] ft260 0003:0403:6030.0010: USB HID v1.11 Device [FTDI FT260] on usb-0000:00:14.0-3.4/input0
[611190.672922] ft260_xfer_status: bus_status 0x20, wakeup
[611190.673173] ft260_xfer_status: bus_status 0x20, clock 100
[611190.673382] ft260_smbus_xfer: smbus size 1
[611190.673388] ft260_i2c_read: rep 0xc2 addr 0x48 len 1 rlen 1 flag 0x6
[611190.703733] ft260_i2c_reset: done
[611190.703749] ft260_smbus_xfer: smbus size 1
[611190.703751] ft260_i2c_read: rep 0xc2 addr 0x49 len 1 rlen 1 flag 0x6
[611190.735902] ft260_i2c_reset: done
[611190.735916] ft260_smbus_xfer: smbus size 1
[611190.735922] ft260_i2c_read: rep 0xc2 addr 0x4a len 1 rlen 1 flag 0x6
[611190.771821] ft260_i2c_reset: done
[611190.771839] ft260_smbus_xfer: smbus size 1
[611190.771846] ft260_i2c_read: rep 0xc2 addr 0x4b len 1 rlen 1 flag 0x6
[611190.803829] ft260_i2c_reset: done
[611190.803846] ft260_smbus_xfer: smbus size 1
[611190.803853] ft260_i2c_read: rep 0xc2 addr 0x4c len 1 rlen 1 flag 0x6
[611190.835829] ft260_i2c_reset: done
[611190.835846] ft260_smbus_xfer: smbus size 1
[611190.835853] ft260_i2c_read: rep 0xc2 addr 0x4d len 1 rlen 1 flag 0x6
[611190.867838] ft260_i2c_reset: done
[611190.867855] ft260_smbus_xfer: smbus size 1
[611190.867862] ft260_i2c_read: rep 0xc2 addr 0x4e len 1 rlen 1 flag 0x6
[611190.899829] ft260_i2c_reset: done
[611190.899847] ft260_smbus_xfer: smbus size 1
[611190.899854] ft260_i2c_read: rep 0xc2 addr 0x4f len 1 rlen 1 flag 0x6
[611190.931830] ft260_i2c_reset: done
[611190.931850] ft260 0003:0403:6030.0010: initialize gpio chip
[611190.931858] ft260 0003:0403:6030.0010: enabled GPIOs: 2c80
[611190.931868] ft260 0003:0403:6030.0010: initialize gpio chip on ft260_0003:0403:6030.0010
[611190.995770] ft260 0003:0403:6030.0011: chip code: 0260 0200
[611190.996113] ft260_get_interface_type: interface:  0x01
[611190.996123] ft260_get_interface_type: chip mode:  0x03
[611190.996127] ft260_get_interface_type: clock_ctl:  0x02
[611190.996130] ft260_get_interface_type: i2c_enable: 0x01
[611190.996133] ft260_get_interface_type: uart_mode:  0x04
[611190.996135] ft260_get_interface_type: gpio2_func: 0x01
[611190.996138] ft260_get_interface_type: gpioA_func: 0x03
[611190.996141] ft260_get_interface_type: gpioG_func: 0x06
[611190.996144] ft260_get_interface_type: wakeup_int: 0x01
[611190.996147] ft260_uart_wakeup_workaraund_enable: activate wakeup workaround
[611190.996395] ft260 0003:0403:6030.0011: registering device /dev/ttyFT0
```

Can you give it a try and provide feedback?

wayne-tung, 2024-05-02:
Thanks, it works good.

I think it's a reasonable change.

But for my project I think it's better to change i2c adapter class to 0.
It will disable i2c auto scan.
Because it will always reset ft260 for every address and I think it's not good.

What do you think about this workaround?

MichaelZaidman, 2024-05-02:
I am glad it helped, and thanks for the feedback!

Disabling the i2C class is a viable alternative. I wanted to propose it to you. However, I was not sure about your application. The talks about removing the HWMON class I2C from I2C drivers have been around for a long time, but very few drivers removed this class up to date. The drawback of this approach is that you will need to instantiate all HWMON-based I2C client devices on the bus by yourself. Also, there is nothing wrong with resetting the I2C controller's internal state on the I2C bus error. I am considering upstream this change and would like to add reported-by and tested-by with your name. If it's OK with you, please send me your e-mail.

Thanks again for reporting the issue.

wayne-tung, 2024-05-02:
Yes, reset the controller on i2c bus error is correct.
I mean it's not good to keep resetting for each address during probing the device.
Because the device work normally, it's a limitation on our driver.

my personal email is: chineweff@gmail.com

MichaelZaidman, 2024-05-02:
> I mean it's not good to keep resetting for each address during probing the device.

It resets the I2C master controller only. Up to 8 resets per LM75 driver gives us several microseconds for all resets. Besides the time wasting, which is negligible now, I cannot think of any other drawback.

wayne-tung, 2024-05-02:
I got your point.
yes, I agree.
Time wasting is the only drawback now.

davetheghost, 2024-05-03:
quick question -- im aware smbus and i2c are very similar.. though i am still perplexed if the ft260 would be able to work with an IO controller like the EC1117. namely a plug and play experience. use case is a laptop palmrest junction board, as the EC1117 has a keyboard encoder, backlight PWM, 2 PS/2 ports along with the i2c for touchpad. *shakes fist at Elan*

apologies for the hijack, figured the question fit best here as there is no actual issue to inquire about. 

ty

MichaelZaidman, 2024-05-03:
> im aware smbus and i2c are very similar.. though i am still perplexed if the ft260 would be able to work with an IO controller like the EC1117.

I cannot see a reason why not.

```$ sudo i2cdetect -F 0
Functionalities implemented by /dev/i2c-0:
I2C                              yes
SMBus Quick Command              no
SMBus Send Byte                  yes
SMBus Receive Byte               yes
SMBus Write Byte                 yes
SMBus Read Byte                  yes
SMBus Write Word                 yes
SMBus Read Word                  yes
SMBus Process Call               no
SMBus Block Write                yes
SMBus Block Read                 yes
SMBus Block Process Call         no
SMBus PEC                        no
I2C Block Write                  yes
I2C Block Read                   yes
```

davetheghost, 2024-05-03:
hmm, what a guy. awesome! meant the ECE1117, however im sure youve gathered that. Ill update once the module arrives -- this could be huge for repurposing things

MichaelZaidman, 2024-05-03:
According to the spec, the ECE1117 uses 4 SMBus commands: Send Byte, Receive Byte, Write Byte, and Read Byte. The ft260 kernel driver supports all. Of course, you must write a user space code to communicate to the ECE1117 via the ft260 kernel driver.

I'm closing this issue as it's been resolved.

closed 2024-05-03 by MichaelZaidman