MichaelZaidman/hid-ft260 #22: Can't compile latest version of the driver on WSL2

mimok, 2024-06-26:
Hi Michael,

I would like to read more than 60 bytes at a time but the version of the driver provided with the linux kernel (v5.15.153.1 from WSL2) seems to be pretty old as i'm facing the error `unsupported read len`.

I tried to compile the latest version of your driver however, i have the following compilation errors:

```error: implicit declaration of function ‘timer_delete_sync’```

```error: initialization of ‘void (*)(struct tty_struct *, struct ktermios *)’ from incompatible pointer type ‘void (*)(struct tty_struct *, const struct ktermios *)’ [-Werror=incompatible-pointer-types]```

Do you have any idea of what is hapenning ?

mimok, 2024-06-26:
i built the latest verison of the kernel available with WSL2 (6.1.21.2) and compiled the driver (commit a096e78ac1ca13206951796daebc03ea630ebe12) and i still have communication problems during write operations:

```console
[ 2276.913629] ft260 0003:0403:6030.0003: ft260_i2c_write: failed with -5
[ 2276.922558] ft260 0003:0403:6030.0003: i2c bus error: 0x2e
[ 2276.925505] ft260 0003:0403:6030.0003: ft260_i2c_write: failed with -5
[ 2276.935484] ft260 0003:0403:6030.0003: i2c bus error: 0x2a
[ 2276.938792] ft260 0003:0403:6030.0003: ft260_i2c_write: failed with -5
``` 

Do you have any suggestions ?

MichaelZaidman, 2024-08-01:
Hey, sorry for the delay in getting back to you. By now, you might have already found a solution. The initial version of the driver was designed for kernel 5.13, while the latest version was developed for kernel 6+. It was expected that it wouldn't compile as is. It's great that you were able to resolve this by switching to kernel 6.1. However, you then encountered I2C bus errors during operation. I2C bus signal integrity is dependent on the custom hardware design, and there could be numerous reasons why it's not working with a specific hardware setup. Since you didn't provide details about your hardware, I'm unable to offer specific help.

closed 2024-09-04 by MichaelZaidman