MichaelZaidman/hid-ft260 #23: HID-FT260: can't build on gcc-14

aayoub13, 2024-07-30:
execution returns error when gcc-14 is being used, caused by incompatible pointers for " tty_operations.write=ft260_uart_write "

aayoub13, 2024-07-30:
Fixed [cafef7f](https://github.com/aayoub13/hid-ft260/commit/cafef7fc01a34c557976530a1265f371f63feb27)

MichaelZaidman, 2024-08-01:
Thanks for the confirmation. I made this change to fix the gcc-13 complaints. See https://github.com/MichaelZaidman/hid-ft260/commit/1b1a2c60ec687d256a6fda2346b93fcefe87fc1d. This change was not propagated to the uart upstream commit, and of course, we need to revert it in this repository also.

aayoub13, 2024-08-01:
Hello, the tty_struct declaration varies from a kernel to another, I distuinguished that on 6.5.0-44-generic, it compiles normally, meanwhile on 6.10.2-arch1-1 it doesn't. so If the mentioned fix is applied, the issue will be resolved on a version and appears on another.

as a workaround I  added 
EXTRA_CFLAGS += -Wno-incompatible-pointer-types
to the Make file  when being built

MichaelZaidman, 2024-08-01:
Correct. It's not because of the GCC but due to the kernel version. I've observed that the data type returned by the tty_operations write function was changed from `int` to `size_t` in this https://www.spinics.net/lists/kernel/msg4894944.html commit, which was merged into kernel v6.6-rc1. Since then, it has always been `size_t`.

While adding the `-Wno-incompatible-pointer-types` switch could be useful in this repository, it will not be accepted into the kernel upstream. Moreover, it might mask real issues during development, which could surface when it's compiled as a kernel built-in driver.

MichaelZaidman, 2025-03-01:
The issue is resolved in https://github.com/MichaelZaidman/hid-ft260/issues/29.

closed 2025-03-01 by MichaelZaidman