MichaelZaidman/hid-ft260 #26: ft260_uart_write() may sleep

jv-oliveira, 2024-10-10:
I tested this driver and got a bug where schedule_timeout() is in the call trace when calling ft260_uart_write(). I checked the [kernel doc for the write tty operation](https://docs.kernel.org/driver-api/tty/tty_driver.html#tty-operations-reference) and it says the function may not sleep.
I managed to work around this with a work queue, but it does not account for errors in the hid output report call. Thus, it is not a solution.

jv-oliveira, 2024-10-10:
I got the bug trying to use this serial port with pppd.
I suppose there would be no problem trying to reproduce the bug, but feel free to reach out for more info on how I got it.

here is my call trace:
```
[   58.290958] BUG: scheduling while atomic: pppd/1085/0x00000404
...
[   58.291067] Call trace:
[   58.291073]  dump_backtrace+0xd8/0x130
[   58.291100]  show_stack+0x18/0x30
[   58.291116]  dump_stack_lvl+0x64/0x80
[   58.291130]  dump_stack+0x18/0x34
[   58.291140]  __schedule_bug+0x58/0x70
[   58.291153]  __schedule+0x534/0x640
[   58.291165]  schedule+0x5c/0xc4
[   58.291176]  schedule_timeout+0x90/0x10c
[   58.291193]  wait_for_completion_timeout+0x78/0x150
[   58.291206]  usb_start_wait_urb+0xec/0x164
[   58.291222]  usb_bulk_msg+0xa0/0x140
[   58.291234]  usb_interrupt_msg+0x10/0x20
[   58.291246]  usbhid_output_report+0x60/0xc4
[   58.291263]  hid_hw_output_report+0x40/0x60
[   58.291275]  ft260_hid_output_report+0x44/0x74 [hid_ft260]
[   58.291319]  ft260_uart_write+0x12c/0x220 [hid_ft260]
[   58.291343]  ppp_async_push+0xe8/0x5cc
[   58.291359]  ppp_async_send+0x48/0x80
[   58.291371]  __ppp_channel_push+0x54/0xe0
[   58.291385]  ppp_channel_push+0x4c/0xa0
[   58.291397]  ppp_write+0x110/0x1a4
```

MichaelZaidman, 2024-10-14:
Hi @jv-oliveira,

Thanks for trying pppd with the hid-ft260 driver. The issue you are observing is not specific to this driver but rather should be common to all uart drivers based on the [linux/drivers/usb/core](https://github.com/torvalds/linux/blob/1d227fcc72223cbdd34d0ce13541cbaab5e0d72f/drivers/usb/core/message.c#L50) code. 
Can you elaborate more on your test setup and the workaround you mentioned?

closed 2025-03-01 by MichaelZaidman