MichaelZaidman/hid-ft260 #35: Large I2C write transaction (~500 bytes) stalls on second 60 byte packet, then resets the I2C controller after `bus_busy` bit stays set for ~100 I2C status packets

chriskeeser, 2025-06-12:
# summary

When attempting to write ~500 bytes to an I2C device, the ft260 fails with the following error in `dmesg`
```
# example command
$ myi2ctool sendfile largewrite.json
...
i2c_adapter:INFO: writing 3 bytes to address 0x28
i2c_adapter:DEBUG: data: [0xC0, 0xE1, 0x01]
i2c_adapter:INFO: writing 497 bytes to address 0x28
i2c_adapter:DEBUG: data: [0x50, 0x00, 0x00, 0x00, ...]
I2C device with 7-bit address 0x28 did not ACK # this message is misleading, the device is ACKing, myi2ctool is reporting the error incorrectly
$ sudo dmesg
...
[12018.556854] ft260 0003:0403:6030.0001: ft260_i2c_write: failed with -5
# this "ft260_i2c_write: failed with -5" error is repeated multiple times
```

A logic analyzer was placed on the I2C lines (as well as some signals from the USB protocol analyzer to see when USB packets were being sent / received) and the I2C device is confirmed to ACK every single byte of each 60 byte packet.  The smaller transactions all go well, but this larger transaction that gets split across multiple 60 byte HID out reports gets stuck on the second packet.

The first trace (RED) is the I2C SCL line
The second trace (Orange) is the I2C SDA line
The third trace (BLUE) indicates the USB Full Speed Start Of Frame (SOF)
The fourth trace (GREEN) is active whenever any valid packet is seen on the I2C bus
The fifth trace (PURPLE) is any OUT packet to endpoint 2 of the FT260 (device 3)
(I apologize if the image quality is a bit potato-ish, I will try to get a better resolution image tomorrow)

[image]

In this image, the "get feature report" packets are shown relative to the I2C transactions as the bottom yellow trace to show the repeated requests to capture the I2C status

[image]

The ft260 was connected to a Beagle 5000v2 USB protocol analyzer, and the problem area was inspected.  (again, apologies for the image quality)

The main story here is that when the first 60 byte OUT packet is sent (with a START condition), the I2C status is checked with the "get report", and it returns that the controller is busy.  The second check shows that the controller is not busy, but the `bus_busy` flag is set.  The second 60 byte OUT transfer is sent (with NO condition, START or STOP), and the I2C status is checked again, with the I2C report indicating the `bus_busy` flag is set (but controller busy not set).  The system then requests the I2C status over and over, each time with the `bus_busy` flag still set (which I think makes sense since the last 60 byte OUT packet indicated no status so the extended I2C transaction is still going) but no other flag set.  Eventually, the system seems to give up and issues an "I2C Reset" packet, and tries again, with the same outcome, ultimately failing with the `dmesg` error shown above.

[image]

My suspicion is that the code that is intended to not check the `bus_busy` flag when you have these combined transactions (hid-ft260.c line 626) is not working as intended and it should be sending the next packet even though the `bus_busy` flag is set.  That's just my hunch, I have not delved deep into the code.

I also suspect that it is possible to replicate this error by writing to an EEPROM with a large number of bytes (> 120).  This would be abusing the EEPROM's write page by overwriting the data in the write buffer over and over and over, but the purpose isn't to actually write the data,  just to have an I2C device ACK on every byte sent of this large combined transaction.  I hope to test that this does indeed replicate the issue soon.

P.s. thank you for your work on this driver, it is well done, and appreciated.  I hope this bug report is sufficient and helpful as well.

## Resources

- Git hash of the commit we are using for the driver: https://github.com/MichaelZaidman/hid-ft260/commit/355aa3eb6d5ee7aebe837de6e073165e4869ec83
- The module is compiled statically into the kernel
- Operating system and kernel information:
```
$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 22.04.5 LTS
Release:        22.04
Codename:       jammy
$ uname -a
Linux orin-0f026b 5.15.148-tegra-g2b8d39c3af72 #13 SMP PREEMPT Thu Apr 24 10:45:15 PDT 2025 aarch64 aarch64 aarch64 GNU/Linux
```
- Hardware running the driver: Nvidia Orin AGX Dev Kit
- Ft260 information
```
$ sudo i2cdetect -l
...
i2c-11  i2c             FT260 usb-i2c bridge                    I2C adapter
$ nvidia@orin-0f026b:/sys/bus/hid/devices/0003:0403:6030.0001$ cat chip_mode
0
clock
100
clock_ctl
2
hid_over_i2c_en
0
i2c_enable
1
power_saving_en
1
pwren_status
1
$ report_descriptor | hexdump -C
00000000  06 00 ff 09 01 a1 02 09  01 95 40 75 08 15 00 26  |..........@u...&|
00000010  ff 00 85 a0 95 3f 09 3a  b1 02 85 a1 95 3f 09 3a  |.....?.:.....?.:|
00000020  b1 02 85 a3 95 3f 09 3a  b1 02 85 a2 95 3f 09 3a  |.....?.:.....?.:|
00000030  b1 02 85 a4 95 3f 09 3a  b1 02 85 b0 95 3c 09 3a  |.....?.:.....<.:|
00000040  b1 02 85 c0 95 3c 09 3a  b1 02 85 c1 95 3c 09 3a  |.....<.:.....<.:|
00000050  b1 02 85 c2 95 04 09 3a  91 02 85 c3 95 3f 09 01  |.......:.....?..|
00000060  91 02 85 d0 95 07 09 01  81 02 09 01 91 02 85 d1  |................|
00000070  95 0b 09 01 81 02 09 01  91 02 85 d2 95 0f 09 01  |................|
00000080  81 02 09 01 91 02 85 d3  95 13 09 01 81 02 09 01  |................|
00000090  91 02 85 d4 95 17 09 01  81 02 09 01 91 02 85 d5  |................|
000000a0  95 1b 09 01 81 02 09 01  91 02 85 d6 95 1f 09 01  |................|
000000b0  81 02 09 01 91 02 85 d7  95 23 09 01 81 02 09 01  |.........#......|
000000c0  91 02 85 d8 95 27 09 01  81 02 09 01 91 02 85 d9  |.....'..........|
000000d0  95 2b 09 01 81 02 09 01  91 02 85 da 95 2f 09 01  |.+.........../..|
000000e0  81 02 09 01 91 02 85 db  95 33 09 01 81 02 09 01  |.........3......|
000000f0  91 02 85 dc 95 37 09 01  81 02 09 01 91 02 85 dd  |.....7..........|
00000100  95 3b 09 01 81 02 09 01  91 02 85 de 95 3f 09 01  |.;...........?..|
00000110  81 02 09 01 91 02 c0                              |.......|
00000117
```

MichaelZaidman, 2025-06-12:
Hi @chriskeeser,

When an I2C clock line (SCL) stays low, it can indicate a few issues, most commonly a clock stretching issue by a slave device or a stuck I2C bus. The slave can hold SCL low to delay transmission while it processes data or prepares for the next transfer.

The I2C slave may not respond correctly or may not release the clock line after a transfer, causing the clock to remain low.

What I2C slave device are you connected to?

Hardware issues like problems with pull-up resistors and wiring may also cause the SCL line to stay low. 

The large write transactions worked on my HW setup in the past. You can see an [example](https://github.com/MichaelZaidman/i2cperf?tab=readme-ov-file#setup-24c512-eeprom-umft260ev1a-evb-200khz-3-hid-ft260-kernel-driver-1-pc-i7-4790k) of writing 128-byte chunks to the EEPROM chip with the [i2cperf](https://github.com/MichaelZaidman/i2cperf) tool by the following command `sudo ./i2cperf -f 2 -o 2 -s 128 -r 0-0xff 13 0x51 -S`. I will retest it on the weekend.

To be able to reproduce the issue on my side, may I ask you to perform the following steps and share the results:
1. Enable debug prints. To do this, you may compile the driver from this repository, unload the built-in driver, and load the built module with the `debug=1` parameter. Since the debug prints will impact the timing, let's see if the issue is reproducible with the latest driver code, without and with debug prints. 
2. To generate a large I2C write transaction, use the `i2cperf` tool with `-f 2 -s 500` flags.

Thanks,
Michael

MichaelZaidman, 2025-06-12:
Hi Chris,

Thank you for your clarification and for the efforts you invested in effectively communicating the issue. I truly appreciate it.

I’ve reproduced the issue on my end and can confirm that you correctly identified its root cause. When I unconditionally passed the value of 0 for bus_busy into the ft260_xfer_status() function, I was able to successfully send 500 bytes.

```$ sudo ./i2cperf -f 2 -o 2 -s 500 -r 0-0x1ff 14 0x51 -S

  Fill block with increment via i2ctransfer by chunks
  -------------------------------------------------------------------
  data rate(bps)  efficiency(%)  data size(B)  total IOs   IO size(B)
  -------------------------------------------------------------------
  71164           87             512           2           500    
```
Of course, this quick fix will cause the combined transactions (write + read) to fail. Hopefully, I will provide a correct fix over the weekend to allow the multi-report writes and combined transactions to operate correctly.

chriskeeser, 2025-06-12:
You are quite welcome.  The effort you put into this weekend project is also appreciated, and a thoughtful bug report is the least I can do.  Thank you for your work on this project!

chriskeeser, 2025-06-12:
I can confirm that this change has addressed the issue I was seeing.  Thank you!

MichaelZaidman, 2025-06-13:
Hi Chris,

If you do not mind, please share your email to add "Reported-by" and "Tested-by" tags to the commit to acknowledge and credit you for this contribution.

Thank you!
Michael

MichaelZaidman, 2025-06-14:
Thanks, Chris! I can now see your email - much appreciated.

No worries at all about the kernel process. Since you reported the issue, I’d like to credit you properly with a Reported-by: tag in the commit message once I upstream the fix. There's nothing else you need to do on your side. Thanks again for your contribution!